rasteri/HIDman #3: Intellimouse features not implemented

rasteri, 2022-09-22:


jarpatus, 2024-05-04:
@rasteri, I am trying to add Intellimouse support and am almost there, but have treoubles on last steps... I wonder if you are willing to help? 

I have implemented PS/2 command buffer so I can detect when driver enables intellimouse (also enabled support for PS/2 resolution and scaling on commands, but those are not that important). I have expanded PS/2 related code to support Z-axis and 4-byte reports. At the moment middle click works under DOS and Win98 and wheel scrolling does work if I fake it (e.g. map mouse Y-axis to wheel like I've done for debugging purposes).

However, I am unable to wrap my head around USB code, reports and stuff. I cannot figure out how I can make USB code to support scrolling. If you are willing to help, could you take a look of this commit on my fork:

https://github.com/rasteri/HIDman/commit/f6b9754f5e3a20176eb96b2270da4323d281251c

In ps2protocol.c I think I was able to do what it takes (case MAP_MOUSE_WHEEL), but in parsedescriptor.c my modifications do nothing (case REPORT_USAGE_WHEEL, do not mind swapped Y and wheel, for debugging only). Any hints?

jarpatus, 2024-05-05:
Hacked together wheel support, it works with ctmouse and under Windows98! But it's a hack as I am parsing wheel input from bare data...

rasteri, 2024-05-06:
Hey I'll look at getting it integrated with the HID report stuff. It's a pretty complicated scheme.

jarpatus, 2024-05-06:
@rasteri, I was able to make it work by adding wheel to standard mouse descriptor... not sure if you can do it like this, I still do not understand HID report descriptors and reports well enough...

https://github.com/rasteri/HIDman/commit/96a871553ec7181b16096e851a17616f6e65b321

jarpatus, 2024-05-06:
Thanks! FYI; I felt andventurous and added support for 5 button intellimouse, this is in it's separate branch intelli5b in my fork. Feel free to try it out if you wish. I had to modify standard mouse descriptor even more.

At least under Win9x IntelliPoint 3.0 software will now detect mouse as Intellimouse Explorer with 5 buttons and buttons really do work. I have no modern machines with PS/2 so I could try it out e.g. with Win11 or so...

rasteri, 2024-05-07:
5-button code works flawlessly in Windows 10. I think we can call this one fixed.

Thanks again for your contribution!

closed 2024-05-07 by rasteri