rasteri/HIDman #13: What enclosure are you using for the original PS/2 version?

sofakng, 2023-06-09:
What enclosure are you using for the original PS/2 version?

It looks like a Hammond case similar to your weeCee case but I'm not sure.

Thanks for the help!

rasteri, 2023-06-09:
It's the 1455D601BK, but the 602 will also work just get whatevers cheaper.

Sorry I'm ignoring vogons, I'm preparing the release of another open source project, Once its done I'll get back to preparing the Hidman 1.0 release

sofakng, 2023-06-09:
No problem and thanks for the reply.  Sorry about the repeated questions.  I was just hoping to order some parts because it takes a few weeks to arrive.

It looks like the only difference between the 601 and 602 are the metal endcaps on the 601 which you aren't using anyways.

Thanks again and I can't wait to see your next project!

closed 2024-05-07 by rasteri