rasteri/HIDman #14: Correct orientation for the LED?

sofakng, 2023-06-20:
Sorry to bother you again...  Can you confirm this is the correct orientation for the LED?  (PS/2 PCB shown;  picture taken through microscope so I apologize for the quality)

[image]

rasteri, 2023-06-21:
yeah I think that's right.
unfortunately the PS/2 version will need some firmware tweaks to work again - the current code only works for AXP...

I should have some time to work on integrating them soon

sofakng, 2023-06-21:
Ahhh, OK.  That probable explains why I was able to flash the chip but it doesn't seem to work.  I'll keep an eye on any updates.  Thanks!

yarodin, 2023-06-24:
Same story here. Waiting fixed firmware for PS/2 version.

rasteri, 2024-05-07:
This has been fixed - latest firmware supports PS/2 version again.

closed 2024-05-07 by rasteri