rasteri/HIDman #15: How to build from source?

DFL1973, 2023-08-31:
Hello, thanks for sharing, but I can´t get the source compiled under Linux or Windows.

I got tons of errors like
main.c:163: syntax error: token -> 'uint32_t' ; column 12
main.c:169: syntax error: token -> 'uint8_t' ; column 11
main.c:183: syntax error: token -> 'uint8_t' ; column 15

Versions:
SDCC -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/TININative/ds400/hc08/s08/stm8 3.5.0 #9253 (Apr  3 2018) (Linux)
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.6.0 #9615 (MINGW32)

Thanks

vtrucco, 2023-09-02:
Fixed after installing SDCC 4.3.0, thanks

DFL1973, 2023-09-02:
Fixed after installing SDCC 4.3.0, thanks

closed 2023-09-02 by DFL1973