rasteri/HIDman #22: Centron does not match Gerbers?

foenixrising, 2024-06-11:
Giving this a shot and sent files into PCBway for fabrication.  Feedback was that components do not align.  Have Gerbers changed but new Centron file hasn't been pushed yet?  Help appreciated, Michael

[image]

rasteri, 2024-06-11:
That centroid file is for the AXP version, not the PS/2 version.

The PS/2 version is missing some features (crystal), I recommend going with the AXP version.

And even then I'd wait till I do an official release. Will be in a couple of weeks hopefully.

foenixrising, 2024-06-11:
Ah, my bad then.  Does the PS/2 version work as of the rev a. board?  (sorry for the bother) if so, I'll send them the updated file, I'm already $100 in on this.  A friend passed me the repo link and I mistakenly took the comments of it being stable but active on face value.  Thanks again.

rasteri, 2024-06-11:
The PS/2 version works somewhat but not on all devices (it has trouble with some hubs).

Did you send them the BOM from the AXP version too? If so they've already priced up the extra AXP components so the AXP won't cost any more.

I would just tell them to make an AXP version instead of the PS/2. Actually works out cheaper (unless you've already bought the enclosure)

foenixrising, 2024-06-11:
I must have sent them a mismatch of Gerber (I'm guessing PS/2) and the BOM and centroid from the AXP.  Can you confirm for me?  You are obviously in the middle of it and I'm lost : )

Also, my needs are simple; using a modern keyboard with no hub and no other USB devices along the bus and want PS/2 keyboard output.  Another use case is a somewhat modern arcade spinner and I'll want PS/2 mouse output.  Alsoalso, I have no idea what AXP stands for?

I don't mean to take up all of your time, if you just tell me which files to use, I'll get the build done and it will either work or not.  I can resubmit in a month or whatever but if this can be made to work, even in the first corner case above, I'll be satisfied.  Thx again.

[Centroid (1).csv](https://github.com/user-attachments/files/15789666/Centroid.1.csv)
[BOM (5).xlsx](https://github.com/user-attachments/files/15789667/BOM.5.xlsx)
[gerbers (1).zip](https://github.com/user-attachments/files/15789668/gerbers.1.zip)

rasteri, 2024-06-11:
Yes, that's the PS/2 version's gerbers with the AXP version's BOM and Centroid.

I consider the PS/2 version to be the "old" version. I never made a BOM or centroid for it as I only ever hand-assembled it.

AXP stands for **A**T/**X**T/**P**S2, it does everything the PS/2 version does and it also has a serial port. It even costs less than the PS/2 version because the enclosure is much cheaper.

Unless it's too late, I would recommend telling PCBWay to build you an AXP version instead. These are the gerbers : https://github.com/rasteri/HIDman/raw/main/hardware/axp/gerbers/gerbers.zip

If you have to have the PS/2 version then I may be able to generate a BOM and centroid

foenixrising, 2024-06-11:
If you've moved on and are only working on the AXP, I'll just send them the new gerber zip and have them remake the board.  The other boards can be used as keychains or something.  Thanks for the feedback, I'll bug you in about a month once they arrive to see where the code is at.  They don't have the PS/2 connectors there so I ordered a bunch from Amazon.  Otherwise, I'm in good shape now, thank you for the help.

closed 2024-06-11 by foenixrising