rasteri/HIDman #32: Problem with cursor keys and some special keys

foenixrising, 2024-08-04:
Thx. for that .hex; flash worked, the module mostly works except for two of what appear to be software/firmware issues:
1. cursor keys, home, insert, etc. do not send any codes to the host; other keys work = all printable, escape (27), return (13), control keys are all normal.  Have tried an old iMAC keyboard that works on a PC and with a newer WASD keyboard that works on a PC or my Macbook
2. the built in menu sends garbage, or at least, some unprintable chars.  Here are the ASCII values of what is sent:

101, 99, 57, 205, 162, 251, 161, 212, 162, 190, 8, 197, 52, 13, 195, 13

If I blindly type "1", I get additional garbage chars.  Any guesses?

rasteri, 2024-08-05:
What type of PC are you connected to?

rasteri, 2024-08-05:
Oh and have you tried doing a HARD factory reset from the menu? (or alternatively check "clear dataflash" when you program the firmware)

rasteri, 2024-08-05:
Yeah there's an option in the menu for "81-key keyboard" that disables a lot of keys. It's intended for REALLY old PCs but if you didnt clear the dataflash it might have defaulted to on. My guess is thats why half the keys don't work.

To be honest I've had problems with the menu corrupting on other non-PC devices, I'm investigating as we speak.

rasteri, 2024-08-05:
Most PS/2 computers will use the Inhibit feature of PS/2 to tell HIDman to delay sending text if it can't process it quickly enough.

foenixrising, 2024-08-05:
I’ll ask the person that did the Verilog if that block included it.  Probably did… just a hunch I had… looks a lot like serial data that is corrupt.  It prints some… garbage then pauses, the I can type a key and the incorrect char echos the more garbage.  I’m unable to ESCape out of that command mode at that ping.  Just wondered if a brief delay between chars would make a diff but I’m not asking for that… will see if I can get anywhere later tonight when I’m home.

rasteri, 2024-08-05:
If you increase PS2_RATE_LIMIT to a few thousand in ps2.h then that will slow everything down. I have to do this on XT PCs.

https://github.com/rasteri/HIDman/blob/7ea815a1be558316a716997f10e8d90ab4194970/firmware/ps2.h#L11

However this may not make a difference. I have other PS/2 hosts that corrupt in a similar way to yours and limiting the rate makes no difference. My guess is it's an incompatibility caused by the inhibit logic but I don't know if it's HIDman's fault or the host's fault.

foenixrising, 2024-08-05:
Any chance you can build one for me with a horse at a delay?  For whatever reason my attempt to build a compilation env and run Make fails.

https://github.com/rasteri/HIDman/blob/7ea815a1be558316a716997f10e8d90ab4194970/firmware/ps2.h#L11

foenixrising, 2024-08-05:
Well, good news/same news.  Good news is clearing out dataflash did the trick; cursor keys and other keys all work now.  'same' news is chars received are still garbled, but are partially discernable now.  Of course typing rate is very slow and I had high hopes but still nothing wonderful; I'll email you a pic of the output.  

So two quick questions; outside of the config not working, is there anything else that might not work properly (as mentioned all keys now work).  Second question, how does one use the 9-pin serial Mouse?  I tried plugging in a USB Mouse on one side and monitored the serial port on the other but didn't see any output?  Thanks.  I have 10 of these built, do you want me to mail you two (not sure how far your hardware design has evolved but if you've only built one or two by hand, I can give you a pair).  Let me know and thanks.

foenixrising, 2024-08-05:
[image]

This is just a simple basic program the does "inkey()" and prints the char with ASCII # and then cprints (which essentially prints the graphic characters to the screen, as you can see with the 13).  Interestingly, all of these chars are normal ascii range values as opposed to the first time I tried it with the 1.1 firmware which output high order ASCII 251, etc.  At this point, I'll see if I can get some info from the person that integrated the PS/2, but if you can respond to my prior message, I'll be all set for now.  Thanks again.

rasteri, 2024-08-06:
The serial mouse should just work, no prep necessary. Does the LED on the front of HIDman blink when you move the mouse/press mouse buttons?

The only other bug I'm aware of is the aforementioned issue with certain other non-PC devices. I have some PS/2-to-USB adapters (i.e. the opposite of HIDman) that don't work, and I'm still to figure out what's going on there. I'll make an effort to investigate this week - maybe if I get them to work it will fix your problem too.

Obviously I've mostly been testing with "Normal" PCs so maybe there's a subtle bug in the firmware that PC's dont mind but other devices have problems with. The logic around inhibits is pretty complicated so it's quite possible.

I actually have 20 HIDmen already made up, so no I don't need any more :) I'll probably be selling a few but not until I do some more testing.

foenixrising, 2024-08-08:
Hey again.  Yes then light flashes w mouse movement.  Trying to scope it through a opto just for fun. Do you know what baud rate and data bits etc for a serial mouse or does it depend on the devices?  Thanks.