rasteri/HIDman #41: Firmware compile error

klyde2278, 2024-10-27:
Hi! I wanted to compile the firmware with BOARD_TYPE = BOARD_MINI, OSC_TYPE = OSC_INTERNAL, and BOARD_OPTIONS = OPT_DEFAULT options activated in the Makefile, under Linux Mint, SDCC 4.0.0, make 4.3-4.1

I never compiled a firmware before. It crashes with the following error.
Do you know what can cause this? Thank you very much for your help. Cheers! Klyde.
```
sdcc -c -V -mmcs51 --model-large --stack-auto --xram-size 0x1600 --xram-loc 0x0200 --code-size 0xEFFF -I/ -DFREQ_SYS=48000000 -DBOARD_MINI -DOPT_DEFAULT -DOSC_INTERNAL  -o build/andyalloc.rel andyalloc.c
+ /usr/bin/sdcpp -nostdinc -Wall -std=c11 -I/ -DFREQ_SYS=48000000 -DBOARD_MINI -DOPT_DEFAULT -DOSC_INTERNAL -obj-ext=.rel -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_MODEL_LARGE -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCC=4_0_0 -D__SDCC_VERSION_MAJOR=4 -D__SDCC_VERSION_MINOR=0 -D__SDCC_VERSION_PATCH=0 -DSDCC=400 -D__SDCC_REVISION=11528 -D__SDCC_mcs51 -D__STDC_NO_COMPLEX__=1 -D__STDC_NO_THREADS__=1 -D__STDC_NO_ATOMICS__=1 -D__STDC_NO_VLA__=1 -D__STDC_ISO_10646__=201409L -D__STDC_UTF_16__=1 -D__STDC_UTF_32__=1 -isystem /usr/bin/../share/sdcc/include/mcs51 -isystem /usr/share/sdcc/include/mcs51 -isystem /usr/bin/../share/sdcc/include -isystem /usr/share/sdcc/include  andyalloc.c 
andyalloc.c:56: error 78: incompatible types
from type 'unsigned-char near* auto'
  to type 'unsigned-char auto'
make: *** [Makefile:75 : build/andyalloc.rel] Error 1
```

rasteri, 2024-10-28:
I'm on SDCC 4.4, try that instead.

Also do a pull before building, I've just merged the hidman-mini and micro code.

klyde2278, 2024-10-28:
Hi! Thank you SDCC 4.4.0 did the trick for the previous source version (1.1.3) and I was able to build a HIDman.bin and HIDman.hex files.

But since you merged serisman PR, I wanted to build again (with the same Makefile options as previously) and now I came across the following error (sorry for being a pain):

+ /usr/local/bin/sdcpp -nostdinc -Wall -std=c11 -I/ -DFREQ_SYS=48000000 -DBOARD_MINI -DOPT_DEFAULT -DOSC_INTERNAL --obj-ext=.rel -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_MODEL_LARGE -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCCCALL=0 -D__SDCC=4_4_4 -D__SDCC_VERSION_MAJOR=4 -D__SDCC_VERSION_MINOR=4 -D__SDCC_VERSION_PATCH=4 -DSDCC=444 -D__SDCC_REVISION=15057 -D__SDCC_mcs51 -D__STDC_NO_COMPLEX__=1 -D__STDC_NO_THREADS__=1 -D__STDC_NO_ATOMICS__=1 -D__STDC_NO_VLA__=1 -D__STDC_ISO_10646__=201409L -D__SIZEOF_FLOAT__=4 -D__SIZEOF_DOUBLE__=4 -D__SDCC_BITINT_MAXWIDTH=64 -isystem /usr/local/bin/../share/sdcc/include/mcs51 -isystem /usr/local/share/sdcc/include/mcs51 -isystem /usr/local/bin/../share/sdcc/include -isystem /usr/local/share/sdcc/include  -xc linkedlist.c 
linkedlist.c:63: warning 283: function declarator with no prototype
linkedlist.c:67: warning 84: 'auto' variable 'list' may be used before initialization
+ /usr/local/bin/sdas8051 -plosgffw build/linkedlist.rel build/linkedlist.asm
make: ***  No rule to build target « build/scancode.rel », needed for « build/hidman.ihx ». Stop.

rasteri, 2024-10-28:
ah I think that's because scancode.c was stored as all-caps (damn Turbo C...). try again now