rasteri/HIDman #47: [Feature Request] Create Mac version with support for ADB and the original Mac 128, 512, Plus keyboard/mouse emulation

aaronnewcomb, 2024-12-02:
Since I mentioned it to Andy, I thought I would open an official issue for tracking this idea. If there were a version of this project that supported ADB and the original Mac formats, it would cover nearly all Macs from the original Mac 128K until the G3 iMac which got rid of the ADB port in favor of USB.

In fact, the form factor of the device could be the same so you could use the same case with a modified faceplate. You would just need to replace one of the PS/2 ports with a 4-pin mini DIN jack and replace the other PS/2 port with an RJ-11 4p4c jack. The 9-pin D-Sub could remain for the old Mac mouse emulation.

There are a number of existing projects that achieve parts of this with dev boards.
adb-usb by gblargg https://github.com/gblargg/adb-usb
ps22plus by n055 https://github.com/n055/ps22plus

rasteri, 2025-01-30:
I think I've tracked down an ADB mac so this may get implemented soon!

rasteri, 2025-01-30:
I still need to find a mac with RJ11 though.