rasteri/HIDman #50: Scancode set 3 support?

techomancer, 2025-01-30:
Hi,
SGI machines require PS/2 scancode set 3 support. So HIDMan doesn't work on them,

rasteri, 2025-01-30:
Oh interesting, I've tried HIDman on some SGI machines and it seemed to work but I guess they're not all the same.

I'll look into it

No0ne, 2025-02-01:
@bstrobl has implemented SGI support in ps2x2pico, maybe you reuse some parts: https://github.com/No0ne/ps2x2pico/pull/39

bstrobel, 2025-02-01:
Let me know if you need anything. Although I don"t have time to code at the moment I can still provide you with traces etc. I traced the communication of a few PS/2 keyboards with my SGI O2 using a logic analyzer to understand the protocol better. BTW the usually referred to documentation regarding scan code set 3 is not quite complete. You really need to trace it.

techomancer, 2025-02-01:
Also, there is definitely variability of PS/2 support between different SGI machines.
I have couple of Filco Majestouch 2 TKL (that has PS/2 backward compatibility), One partially works with Indigo 2, one doesn't at all.
But both work fine with O2. Haven't tried them yet with Indy or Octane.

bstrobel, 2025-02-03:
To be honest seeing the variety of problems regarding normal modern USB keyboards with normal USB communication I wouldn't put too much trust in the claimed PS/2 compatibility of them. Your really should test your SGIs with original 1990s and 2000s PS/2 keyboards. If they work you could trace the communication with a cheap $10 logic analyzer and use the code of ps2x2pico as a basis for implementing scancode set 3 support in HIDman. I guess the changes will be minor. I can support you with that a bit if you want.

Or you could just build a ps2x2pico yourself (investment of about $10 to $15) and try it out. Maybe (I even dare to say probably) it works out of the box.

techomancer, 2025-02-03:
I probably will, I ordered some level shifters, I also have DreamSourceLab logic analyzer so I can hook it up.
I just have too many projects pending ...

next68k, 2025-02-15:
I would like to help. Trying to get a Leopold FC900R working on my SGI O2.  I get very erratic behavior with this keyboard and a PS/2 adapter. Like keys randomly stop working and such. Board seems to work fine in USB mode on a mac or pc.  I don't know if it is a scancode issue or I need a different PS/2 adapter.