rasteri/HIDman #53: Various Wireless Mice X, Y, Z Axis bug

writeguide6667, 2025-02-18:
I have tried all of these wireless mice connected to the HIDman and they seem to work somewhat, but all have erratic movement and or always lock the mouse cursor position to the bottom of the screen. the only mouse that i've tried that is fully functional in my possession is a

[g304.txt](https://github.com/user-attachments/files/18853525/g304.txt)
[m187.txt](https://github.com/user-attachments/files/18853526/m187.txt)
[yichip.txt](https://github.com/user-attachments/files/18853524/yichip.txt)

 wired USB Logitech B100 Mouse. The offending wireless mice are as follows;

1. Logitech G304 Lightspeed

2. Logitech M187 Mini Wireless Mouse

3. YICHIP Wireless Device "Generic Wireless Black Mouse"

writeguide6667, 2025-02-19:
with the current firmware, the G304 with advance USB mode off don't lock to the bottom of screen, but has that wierd jumping/resetting position erratic movement mentioned earlier.

YICHIP works properly with advanced USB mode off. the only strange issue is that right clicking the mouse button just makes the cursor drift to the right instead of opening the right click menu.

all three of these mouse do lock to the bottom of screen when enabling advanced mode USB.
this was tested on Windows 98 SE and mageia 9 linux.

with that being said, i'll try the new firmware to see if the m187 changes behavior with advanced USB mode ON or OFF.

writeguide6667, 2025-02-19:
i tested the new firmware. Both the M187 and Yichip wireless mice now work properly with advanced USB mode both ON and OFF.

The G304 still has problems with advanced USB mode ON with a slight behavior change with mouse cursor no longer locking to the bottom but instead flickering all over the desktop. it seems work okay with advanced USB mode off, however i'm still getting random left clicks and erratic cursor movement on occasion. this seems to happen when i'm just only moving the mouse around the desktop.

in total of wireless mice, i'm satisfied with the other two working fine.

rasteri, 2025-03-03:
Yeah the G304 is sending what are essentially corrupted packets when it's not running in advanced mode. It may not be possible to get it running in non-advanced mode.

If you have a chance, could you send me a HID log of the G304 in advanced mode? Plug it in and then move it around a bit and press the buttons some.

writeguide6667, 2025-03-04:
here is the HID log for the G304 with advanced USB mode on.

[G304.TXT](https://github.com/user-attachments/files/19078073/G304.TXT)

rasteri, 2025-03-08:
Aha, the G304 sends 12-bit X and Y coordinates, which I haven't tested fully yet.

I've bought a G304, found one on ebay for a fiver :)

writeguide6667, 2025-03-09:
the lightspeed G304 wireless mouse; when getting one used, you will probably want to connect to usb on a regular PC to test for accidental double clicking because the mouse may have worn out switches that may interfere with your tests.

rasteri, 2025-03-25:
So I bought a G304 and also get glitchy output. But it turns out the G304 I bought is fake - its vendor ID indicates it's made by Yichip.

So I looked up the vendor ID of your G304, and it's made by SEMITEK. 

Can your G304 connect to Logitech G-Hub software? If not I suspect it's fake too.

writeguide6667, 2025-03-25:
So i tried the Logitech G-Hub suite, and yeah... it can't find the mouse, wow. 

whats even worst is i did an usb device listing under linux and it shows up as an "SHARKOON Technologies GmbH 2.4GHz Wireless rechargeable vertical mouse [More&Better]" with usb ID 1ea7:0064

it indeed is a fake, which to my surprise because is quite humorous since i've just got to use this mouse as i had it sitting around for quite awhile in it's still new packaging bought off from amazon. Woops.

writeguide6667, 2025-03-25:
i've must of had real g304 at one point because i've bought this one to replace the old one because of the annoying double clicking, i never even thought of ever using their software as i assumed it was just bloatware.

rasteri, 2025-03-25:
what a strange turn of events! haha.

My mouse seems to be even more glitchy than yours, it knocks itself out of boot mode every time you rightclick (but not leftclick). bizarre.

I'm still gonna try to get hidman working with it though. If Windows can use it then hidman should be able to too.

writeguide6667, 2025-03-26:
all of these mice seem to work fine in both windows and linux when connected directly to the PC via USB connection.

it's just in the hidman where things get whonky. this unfortunately is going to get tricky when we have a bunch of devices pretending to be some something they aren't i.e. fake mice.

writeguide6667, 2025-05-20:
Hey again, just letting you know with the newly released 1.1.5 firmware from today has solve all the above issues, even all of my fake chinese USB mouse are now fully functional. i had ended up buying a logitech g305 from Micro center just last week to see if they are also vulnerable to selling fake branded computer peripherals and as it turns out the mouse i got was a real Logitech mouse. It even reports as such under USB HID info under Linux. 

Now that all my computer mice work, The Micro center purchase became pointless. oh well... at least i got a free 128 SSD with the purchase.

So with that being said, i guess this issue can be closed unless someone else finds the exact same problems with the current HIDman firmware.

closed 2025-05-21 by rasteri