rasteri/HIDman #55: More widespread case for AXP?

yarodin, 2025-03-30:
Is there analogue for this https://lincolnbinns.com/e-case-a-40mm-black.html  case for AXP on aliexpress etc?
It's  totally impossible to gets one at my location.

rasteri, 2025-03-30:
Evatron sells them under a different name if that helps at all : https://evatron.com/enclosures/aluminium-enclosures/aluminium-enclosures-re-series/reab40-aluminium-extrusion/

I can't find any on aliexpress though

nizmow, 2025-04-25:
I found these on element14 which fit perfectly: https://au.element14.com/multicomp/mcreas40/case-aluminium-40x63-5x30mm/dp/1549964

Not black though!

tomtebloss, 2026-02-07:
I sliced the 80mm step-file that Lincoln Binns have on the website and what is used as the original enclosure in aluminium. This file is 40mm [eas40.zip](https://github.com/user-attachments/files/25153489/eas40.zip) and I ordered mine from JLC3DP.com in SLA and JLC Black Resin. looks great.