rasteri/HIDman #56: Problem with PCB/Assembly (report from NextPCB)

yarodin, 2025-04-21:
I'm trying to order assembled Hidman AXP from NextPCB (the only service available at my location).
I took gerbers, centroid and BOM files from [https://github.com/rasteri/HIDman/tree/main/hardware/axp](https://github.com/rasteri/HIDman/tree/main/hardware/axp).
But suddenly I got error report from Design for Assembly (DFA) and I don't understand what's wrong and what should I do next.

[NextPCBDFAReportTR63009AA1.xlsx](https://github.com/user-attachments/files/19832380/NextPCBDFAReportTR63009AA1.xlsx)

[BOM_4_B-R63009AA1.xlsx](https://github.com/user-attachments/files/19832381/BOM_4_B-R63009AA1.xlsx)

yarodin, 2025-04-24:
Сommunication with PFA Team of NextPCB is continuing. According to their information the main problem with J1,J3 DIN-6 conn.

"For J3,J1, the difference and the dimension of the holes between the Gerber and datasheet is too big, so there is a big risk that the components can't be inserted in"

[image]

[image]

[image]

RoberMC, 2025-05-03:
That's because the footprint does not match the part, it seems it is for another kind of mini-din with more pins, but the correct Mini-Din 6 part should fit with some manual force, I had that issue myself. Usually manufacturers will complain about that if you tell them to asemble the through hole parts. It's better to assemble that part manually by yourself.

yarodin, 2025-05-03:
Yes, you are right. In BOM file DIN-6 connector with six pin on the PCB part: [https://www.lcsc.com/datasheet/lcsc_datasheet_2409302302_HOOYA-DIN-603_C7428703.pdf](HOOYA DIN-603), 
whereas PCB designed for a slightly other type of connector (eight pins on th PCB part)

May be it will be a good idea to adjust Hidman PCB to exactly DIN-6 connectror from project BOM file?

In sum I ordered assembled devices without mounted DIN-6 connector from NextPCB. Interesting, but PCBWay sent to production exactly same files (gerbers, BOM, centroids) without any question. I will look to result ;)

rasteri, 2025-05-03:
Hey, I'll take a look at modifying the PCB to fit the footprint better. I guess I just never noticed the problem because the parts fit ok.

rasteri, 2025-05-03:
I used the footprint for 8 pin because I thought they were the same and I had a previous project using 8pin devices. Ah well, laziness fails me again...