rasteri/HIDman #57: 8bitdo retro bluetooth/2.4ghz keyboard works connected with USB but not wirelessly

keenerb, 2025-05-01:
Attached as USB it works perfectly.

Attached as wireless, I never see a link light on the 8bitdo keyboard but the hidman LED flashes as if it's receiving data, and the HID capture shows some sort of event intercepted but never see any characters entered.

[ASDF.txt](https://github.com/user-attachments/files/20003583/ASDF.txt)

The capture was started with device attached as USB, at "renumerating all ports" I had inserted the wireless dongle and switched to 2.4ghz mode.  I unplugged the USB cable around line 100 in the log output.

I hit the A key four or five times, and then the B and C keys once each.

rasteri, 2025-05-13:
Two things - 

- Try the latest beta firmware (v1.15i) - https://github.com/rasteri/HIDman/releases/tag/v1.1.5i
- If that doesn't work, send me another log but this time unplug the wired keyboard before plugging the wireless in

keenerb, 2025-05-14:
[8BITDO.TXT](https://github.com/user-attachments/files/20208577/8BITDO.TXT)

I started debug mode with a wired keyboard and then unplugged it, and plugged in the 8bitdo wirelessly and typed A - J (or something along those line.)

When I reconnected the original keyboard to save the document, it would not work either.  I had to unplug the hidman from the computer and then re-connect to be able to save the document.

rasteri, 2025-05-14:
Yeah there's a bug in 1.1.5i where the log mode doesn't exit properly....

I'll look over your results.

keenerb, 2025-05-14:
Thanks.  It works fine using USB cable so it's not exactly a critical issue.

rasteri, 2025-05-17:
It seems from the logs that the simple USB mode on the dongle doesn't work. Try turning on "Advanced USB" in BOTH the Keyboard AND Mouse menus (a lot of devices don't like only one being turned on). See if that improves anything

vanfanel, 2025-05-20:
> It seems from the logs that the simple USB mode on the dongle doesn't work. Try turning on "Advanced USB" in BOTH the Keyboard AND Mouse menus (a lot of devices don't like only one being turned on). See if that improves anything

Hey Rasteri!
Another 8BitDo Retro Keyboard user here.
The keyboard works fine after activating the "Advanced USB" option.
If that's a valid solution (well, it works fine here) then this issue can be closed I believe.

rasteri, 2025-05-21:
I might contact 8bitdo and point out the bugs in their keyboard... not sure if there's anything they can do to update them now they're released though.

vanfanel, 2025-05-21:
> I might contact 8bitdo and point out the bugs in their keyboard... not sure if there's anything they can do to update them now they're released though.

Well, these 8bitdo keyboards can have their firmware updated (with very poor quality Windows software instead of proper fwupd support, but well...) so yes, please do contact them on the issues. I tried but I lack patience with their poor support denying GNU/Linux existence.