rasteri/HIDman #60: Tinyllama 2 firmware update

GuerreroNinja, 2025-05-16:
After sucessfully updating the Hidman firmware to be compatible with KVMs, I also tried to update the CH559 of my Tinyllama2 with the latest hidman firmare. No luck this time, and I tried all pre-release and stable firmwares released in this repo. The updating instructions for the Tinyllama2 are basically the same as the Hidman, so I think I am doing right the procedure. 

I checked the original Tinyllama2 firmware bin file, and seems it is 2 years old, so maybe the new hidman firmware are not compatible "as is" with the Tinyllama2?

rasteri, 2025-05-17:
As far as I know the firmware should be compatible. Does it just not work at all, or just not with kvms?

GuerreroNinja, 2025-05-17:
It does not work at all.

rasteri, 2025-05-17:
Ahh I know what the issue is. Tinyllama2 is based on an earlier version of HIDman without the crystal oscillator.

I've compiled the latest HIDman firmware for the old hardware, it should work but it's possible KVMs might not (the crystal adds stability)

https://github.com/rasteri/HIDman/releases/tag/v1.1.5j

GuerreroNinja, 2025-05-17:
Thank you! Please let me check it, I will provide some feedback within the hour.

GuerreroNinja, 2025-05-17:
Ok, Tinyllama updated with the 1.1.5j firmware and I can confirm it works both direct connection and through my KVM without issues. Thank you!!

GuerreroNinja, 2025-05-17:
UPDATE: After intensive testing, unfortunately it is not stable. The keyboard/mouse connected to the KVM fail at boot in most cases and I have to unplug and plug again several times the USB cable from the Tinyllama until the keyboard/mouse is detected. Once detected, it works normally without problems (though one time it disconnected by no reason while typing in Edit). Overall, the patch adds the KVM functionality to Tinyllama, but it is not stable.

rasteri, 2025-05-17:
Yeah if your KVM works with the crystal hidman but not the tinyllama2 then this is likely a hardware problem.  

I believe tinyllama3 is going to have a crystal on it...

GuerreroNinja, 2025-05-18:
Then it is a hardware limitation for sure. The Tinyllama 3 poject is still in development, but the ITX Llama includes PS/2 ports, so you can use the external Hidman hardware or the included USB through RP2040.

I am closing then this issue, thank you for your help Rasteri!

closed 2025-05-18 by GuerreroNinja