rasteri/HIDman #63: Feature request:  Basic joystick keystroke remapping via built-in menu

keenerb, 2025-05-25:
It would be nice to be able to access a menu under 3. Game that would prompt for something along these lines:

Up: <accepts single keypress>
Down: <accepts single keypress>
Left: <accepts single keypress>
Right: <accepts single keypress>

Button 1: <accepts single keypress>
Button 2: <accepts single keypress>

So you could access the menu and do something like:

Up: w
Down: a
Left: s
Right: d

Button 1: ctrl
Button 2: alt

and have the joystick mapped to those keyboard keypresses.

rasteri, 2025-05-27:
Yes this is a feature that has been on my mind for some time. Essentially all the back-end code is there it just needs the menu to be coded.

RoberMC, 2025-06-27:
It would be better if Hidman could be accessed from the computer using serial port to configure the remaps. This way one could launch a game with a batch file sending the remap and then launching the game's executable. I haven't checked the diagram so i don't know if serial receive pins are wired to the MCU though.