rasteri/HIDman #68: Only one USB cable out from KVM - will HIDman find mouse AND keyboard?

TimRegeant, 2025-06-25:
Hi, my KVM takes a console side USB mouse and keyboard but outputs only to one USB cable which is connected to each host side computer.  My computers automatically find the mouse and keyboard packets I guess and everything works good.

For my PS/2 computer I am interested in using HIDMAN, but I see it takes a possible 2 USB inputs.

My question is:  Will HIDman be able to see both mouse and keyboard packets on just the one USB cable from my KVM and output them on the appropriate PS/2 port?

TimRegeant, 2025-06-29:
Btw, is there any similar project that take a single usb cable and split it back to 2 usb outputs - mouse and keyboard?

The reason I ask is as I said my kvm combines the mouse and keyboard cables into 1 usb.  But I have a computer that needs specifically to have mouse on a certain usb port and keyboard in another specfic port.

benbaker76, 2025-11-18:
The StarTech KVM I have uses an ActionStar chip. Here's a summary of the Device Descriptor

```
Information for device USB Mouse (VID=0x0835 PID=0x1411):

Connection Information:
------------------------------
Device current bus speed: LowSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x000B
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x0835	idVendor
0x1411	idProduct
0x0100	bcdDevice
0x03	iManufacturer   "DP KVM"
0x02	iProduct   "USB Mouse"
0x00	iSerialNumber
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x003B	wTotalLength   (59 bytes)
0x02	bNumInterfaces
0x01	bConfigurationValue
0x03	iConfiguration   "DP KVM"
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x01	iInterface   "USB Keyboard"

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0041	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x0A	bInterval         (10 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x02	iInterface   "USB Mouse"

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x007B	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x0A	bInterval         (10 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x03   0x0409  "DP KVM"
0x02   0x0409  "USB Mouse"
0x01   0x0409  "USB Keyboard"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
Generic USB Hub
USB Mouse (VID=0x0835 PID=0x1411) Port: 4

Running on: Windows 10 or greater (Build Version 26200)

Brought to you by TDD v2.12.0, Aug  2 2019, 16:35:05
```

rasteri, 2026-01-09:
USB Hub and KVM support is much improved in latest beta - https://github.com/rasteri/HIDman/releases/tag/v1.1.7beta1