rasteri/HIDman #71: USB Hub issues and diagnose, It's "almost" working.

RoberMC, 2025-07-11:
Hello, first congratulations for this amazing device and your great work, I got it yesterday and I am having Issues with a pretty common USB HUB, Here's what I observed and I hope my findings help making it more stable.

The USB HUB is based on the Terminus FE2.1 chip (USB2.0, 7 ports)

- When the HUB is connected the remaining native hidman port starts behaving erratically, it skips key presses or repeat keys randomly
- Devices connected to the HUB do not work, nothing is registered, blue light does not blink.
- Enter the Menu - Advanced - Log HID Data, do a Rescan.
- After the rescan completes everything connected to the HUB start working 100% perfectly while the native HIDMAN port keeps behaving erratically
- If the HUB is unplugged the native port starts to work fine again.
- If you plug again the HUB, there's 80% chance devices connected to it will not be recognized, after a rescan they will be detected again.
- Native Hidman USB ports will never work fine as long as the hub is connected to the other port.

The devices I have connected are:
HIDMAN USB1: Noname USB mechanical Keyboard
HIDMAN USB2: FE2.1 HUB ---> Logitech Unifying Receiver (Logitech K400 Plus keybd+trackpad)

I attached the log when doing a rescan and the devices on the hub pass from not being recognized to be recognized.

I hope this information is helpful, thank you for the hard work you put into this.

[HIDMAN_LOG1.TXT](https://github.com/user-attachments/files/21184848/HIDMAN_LOG1.TXT)

jarpatus, 2026-02-14:
Interestingly enough, I managed to get chained hubs to ALMOST work when I connect my final hub to certain port on my Dell display (sadly, broken one, it sometimes is fiddly as connector is physically broken). But not fully, it works for few seconds, after led on hub blinks, devices are disconnected and soon after it works again.

When connected like this, it almost works, when connected in any other way it does not:

[image]

rasteri, 2026-02-18:
Oh - hidman micro is based on older HIDman hardware, which doesn't have an oscillator on it. It will have unavoidable problems with hubs I'm afraid.

It's possible the full hidman will work on your setup, or micro might if it was modified to have an oscillator on it