rasteri/HIDman #75: Onikuma Gaming Mouse CW917 compatibility issues.

GuruSR, 2025-09-03:
Firmware 1.1.6.
Mouse configuration is No for both Adv and Intellimouse.  If either are Yes, the mouse fails to do basic work (mouse buttons work though).

LRM and movement work with Adv and Intellimouse both at No.

This is a basic 5 button mouse (gaming'ish), 3600/2400/1800/1200 DPI.  LRM, wheel and 2 side buttons.

Their website is:  www.onikumagaming.com

[MouseInfo.txt](https://github.com/user-attachments/files/22118911/MouseInfo.txt)

This is going to be plugged into a PS/2 KVM to allow me to use other mice.  The above mouse does see some lag if you move it around a lot (fast), visually can see when I stop the mouse, the pointer is still moving.

I have an old usb to PS/2 adapter (from a Microsoft PS/2 mouse kit), that this mouse does work directly with on the KVM, but only the mouse buttons and movement (no wheel).  I didn't see any lag this way.

rasteri, 2026-01-02:
sorry for the delay, yeah mine works as long as I have both advanced keyboard and advanced mouse enabled.

GuruSR, 2026-01-31:
I have it running with everything set On (Advanced Mouse and Keyboard), but I'm seeing really bad lag, is there any way to improve that so it doesn't lag?

TXDokie1, 2026-10-28:
w